VeriSilicon/TIM-VX #424: Tile compute error

MESeraph, 2022-07-04:
I have a onnx case [tile](https://github.com/VeriSilicon/TIM-VX/files/9037527/tile.zip) UT that can passed  on onnxruntime.

I add a UT for Tile based on the onnx case : (need put the [tile](https://github.com/VeriSilicon/TIM-VX/files/9037527/tile.zip) folder under **host_build/sr/tim**)
tile_test.cc
```
#include "tim/vx/context.h"
#include "tim/vx/graph.h"
#include "tim/vx/ops/tile.h"
#include <iostream>
#include <fstream>
#include <numeric>
#include "gtest/gtest.h"
#include "test_utils.h"

union float32_bits {
  unsigned int u;
  float f;
};

struct Float16 {
  uint16_t val;

  Float16() : val(0) {}
  explicit Float16(uint16_t x) : val(x) {}
  //explicit MLFloat16(float f);

  float ToFloat() const
  {
    const float32_bits magic = { 113 << 23 };
    const unsigned int shifted_exp = 0x7c00 << 13; // exponent mask after shift
    float32_bits o;

    o.u = (val & 0x7fff) << 13;             // exponent/mantissa bits
    unsigned int exp = shifted_exp & o.u;   // just the exponent
    o.u += (127 - 15) << 23;                // exponent adjust

    // handle exponent special cases
    if (exp == shifted_exp) {     // Inf/NaN?
        o.u += (128 - 16) << 23;    // extra exp adjust
    } else if (exp == 0) {        // Zero/Denormal?
        o.u += 1 << 23;             // extra exp adjust
        o.f -= magic.f;             // renormalize
    }

    o.u |= (val & 0x8000) << 16;    // sign bit
    return o.f;
  }

  operator float() const {
    return ToFloat();
  }
};

TEST(Tile, shape_6_2_11_float_multiples_2_3_1) {
    auto ctx = tim::vx::Context::Create();
    auto graph = ctx->CreateGraph();

    tim::vx::ShapeType input_shape({6, 2, 11});
    tim::vx::ShapeType output_shape({12, 6, 11});
    tim::vx::TensorSpec input_spec(tim::vx::DataType::FLOAT16,
        input_shape, tim::vx::TensorAttribute::INPUT);
    tim::vx::TensorSpec output_spec(tim::vx::DataType::FLOAT16,
        output_shape, tim::vx::TensorAttribute::OUTPUT);

    auto input_tensor = graph->CreateTensor(input_spec);
    auto output_tensor = graph->CreateTensor(output_spec);

    
    auto in_count = std::accumulate(input_shape.begin(),
                                        input_shape.end(), 1, std::multiplies<int64_t>());
    u_int16_t in_data[in_count] = {0};
    std::ifstream in("tile/test_data_set0/input_0.bin", std::ios::in | std::ios::binary);
    if(in.is_open())
    {
        in.read((char *) &in_data, sizeof(in_data)*sizeof(float));
        in.close();
    }
    else
    {
        std::cout<<"open input_0.bin fail!"<<std::endl;
    }
    

    auto out_count = std::accumulate(output_shape.begin(),
                                        output_shape.end(), 1, std::multiplies<int64_t>());
    u_int16_t golden[out_count] = {0};
    std::vector<float> golden_float(out_count);
    std::ifstream out("tile/test_data_set0/output_0.bin", std::ios::in | std::ios::binary);
    if(out.is_open())
    {
        out.read((char *) &golden, sizeof(golden)*sizeof(u_int16_t));
        for(auto i=0; i<out_count; i++)
        {   
           golden_float[i] = Float16(golden[i]).ToFloat();
        }
        out.close();
    }
    else
    {
        std::cout<<"open output_0.bin fail!"<<std::endl;
    }

    EXPECT_TRUE(input_tensor->CopyDataToTensor(
        in_data, sizeof(in_data) * sizeof(u_int16_t)));

    std::vector<int32_t> multiples = {2, 3, 1};
    auto op = graph->CreateOperation<tim::vx::ops::Tile>(multiples);
    (*op).BindInputs({input_tensor}).BindOutputs({output_tensor});

    EXPECT_TRUE(graph->Compile());
    EXPECT_TRUE(graph->Run());

    std::vector<u_int16_t> output(sizeof(golden)*sizeof(u_int16_t));
    std::vector<float> output_float(out_count);
    EXPECT_TRUE(output_tensor->CopyDataFromTensor(output.data()));
    for(auto i=0; i<out_count; i++)
    {   
        output_float[i] = Float16(output[i]).ToFloat();
    }
    EXPECT_TRUE(ArraysMatch(golden_float, output_float,1e-4f));
}
```
After running, results is zero all.

sunshinemyson, 2022-07-06:
Can you share our driver log with VIV_VX_DEBUG_LEVEL=1 for this issue? Thanks.

MESeraph, 2022-07-06:
@sunshinemyson 
[tile.log](https://github.com/VeriSilicon/TIM-VX/files/9052180/tile.log)

xuke537, 2022-07-13:
Tile operation multiples parameter setting issue，we will fix it.

MESeraph, 2022-07-21:
Will the issue be fixed on the next prebuild sdk release?

xuke537, 2022-07-21:
No, the next prebuild sdk release is on the way. There's not enough time.

MESeraph, 2022-09-23:
Verify that is fixed.

closed 2022-09-23 by MESeraph