VeriSilicon/TIM-VX #427: "*.tflite" to "*.tmfile" conversion issue.

kprastey, 2022-07-06:
Need to run custom mobilenetV2 based "*.tflite" model on the Khadas-VIM3's NPU.

Tried converting a custom mobilenetV2 based classifier model from "*.tflite" to "*.tmfile", but faced the following error.

[image]

It seems like one of the operations (i.e. DEQUANTIZE) isn't supported.

also tried "*.tflite" -> "*.onnx" ->"*.tmfile". which gave similar errors.

[image]

Looking forward to having technical assistance from your end regarding running custom "*.tflite" models having operations like DEQUANTIZE, QUANTIZE, etc. on the khadas VIM3's NPU.

sunshinemyson, 2022-07-06:
Sorry, We don't have knowledge about this tool. We can support tflite model directly with tensorflowlite + vx-delegate, you can try this approach.

kprastey, 2022-07-11:
Hey @sunshinemyson,

- It'll be great if you could guide with some documented steps or provide any reference material to help use "tensorflowlite + vx-delegate", especially on the Khadas VIM-3 platform.
- If that's not possible then any platform-agnostic references would also be appreciated.

sunshinemyson, 2022-07-18:
@kprastey ,

https://github.com/VeriSilicon/tflite-vx-delegate Can you to work out by follow the README.md firstly?

closed 2022-09-28 by sunshinemyson