VeriSilicon/TIM-VX #485: Can I use prebuilt-sdk of x86_64_linux to test OpenVX Nodes example ?

michaelnguyen11, 2022-09-19:
Hi admin,

I got OpenVX Node bugs at NPU SDK version 6.4.3, and know that the prebuilt-sdk for linux version is 6.4.11.
Can I use the prebuilt-sdk to check whether with the newer version, the issue is fixed or not ? If yes, so :

I created a new folder under sample folder and added example, could build and run it. However, the display result is totally empty. Attached the CMakeLists.txt file for example code

`set(OPENVX_INCLUDES "/home/hiep/Desktop/hiep_test/TIM-VX/prebuilt-sdk/x86_64_linux/include/VX")

add_library(OpenVX_LIBS SHARED IMPORTED)
set_property(TARGET OpenVX_LIBS PROPERTY IMPORTED_LOCATION "~/Desktop/TIM-VX/prebuilt-sdk/x86_64_linux/lib/libOpenVX.so")
...
target_link_libraries(${TARGET_NAME} PRIVATE tim-vx ${OpenCV_LIBS} ${OpenVX_LIBS})
target_include_directories(${TARGET_NAME} PRIVATE
    ${CMAKE_CURRENT_SOURCE_DIR}
    ${PROJECT_SOURCE_DIR}/include
    ${OPENVX_INCLUDES}
)`

Many thanks in advance !

sunshinemyson, 2022-09-19:
@michaelnguyen11 ,

Yes, you can verify it with latest version. I cannot find out any clue with your makefile. Please share full log with env VIV_VX_DEBUG_LEVEL=1 & VSI_NN_LOG_LEVEL=5, and source code of your test program if it is a unit test.

Thanks

michaelnguyen11, 2022-09-19:
Hi @sunshinemyson ,

Thank you for your informaton.

Please check the attached file for the test program:
[debug_openvx_nodes.zip](https://github.com/VeriSilicon/TIM-VX/files/9600869/debug_openvx_nodes.zip)

Unzip and build it as the following:
```
make
cd build
./openvx_bug_test ../PETS09-S1-L1-View001.avi
```
Attached full log with env VIV_VX_DEBUG_LEVEL=1 & VSI_NN_LOG_LEVEL=5 , please help to check it : [full_log_with_env.txt](https://github.com/VeriSilicon/TIM-VX/files/9600867/full_log_with_env.txt)

michaelnguyen11, 2022-09-22:
Hi @sunshinemyson ,

Thank you for the information, hope to see your response soon.

sunshinemyson, 2022-09-26:
@michaelnguyen11 ,

We reproduced your issue internally and working on the debug.

Thanks.

sunshinemyson, 2022-10-08:
Internal bug fixed. Will update prebuild later this month.

sunshinemyson, 2022-11-02:
[image]
Verified with 22Q3 prebuilt sdk.

closed 2022-11-02 by sunshinemyson