VeriSilicon/TIM-VX #529: PLT Section missing in galkore

kventinel, 2022-12-07:
Hello, I try to use galcore.ko from libs in release https://github.com/VeriSilicon/TIM-VX/releases/tag/v1.1.42, but have next output:
```
/alexrak/vim3l_aarch64/lib # insmod galcore.ko
insmod: can't insert 'galcore.ko': invalid module format
/alexrak/vim3l_aarch64/lib # dmesg
[1128771.137591] galcore: module PLT section(s) missing
[1128771.161401] galcore: module PLT section(s) missing
```

Can you help me to find out from error with PLT section missing?

sunshinemyson, 2022-12-13:
@kventinel ,

Did you tried contact khadas for help?  I think this maybe a compatible issue between the kernel driver and your BSP.

kventinel, 2022-12-14:
I have found already, that problem was in 32-bit userspace on my device. But currently I need to build Tim-vx from source, but don't know where I can find [prebuilt SDK](https://github.com/VeriSilicon/TIM-VX/tree/main/prebuilt-sdk) for arm7. Can you help me with that?

sunshinemyson, 2022-12-16:
Hi kventinel,

Do you have the our SDK from khadas? If you have them, you can build tim-vx from source by set EXTERNAL_VIV_SDK properly.

kventinel, 2022-12-20:
Nope.

closed 2023-10-11 by sunshinemyson