VeriSilicon/TIM-VX #551: EmbeddingLookup.shape_2_2_2_3_Uint8QuantizedLUT test fails with Segmentation fault

robert-kalmar, 2023-01-31:
The test EmbeddingLookup.shape_2_2_2_3_Uint8QuantizedLUT fails with Segmentation fault on VIP8000 (i.mx8MPlus).
Configuration: 
- GPU Driver 6.4.11
- standalone OVXLIB - version 1.1.67 (patch release patch_6.4.11_20221222)

```
[----------] 2 tests from EmbeddingLookup
[ RUN      ] EmbeddingLookup.shape_2_5_int32LUT
[       OK ] EmbeddingLookup.shape_2_5_int32LUT (19 ms)
[ RUN      ] EmbeddingLookup.shape_2_2_2_3_Uint8QuantizedLUT
Segmentation fault
```

sunshinemyson, 2023-02-06:
@robert-kalmar ,

You can apply patch from tim-vx ac4517b5c1c8e234317f5759c9d83bf0cbc7b163 to standalone ovxlib.

robert-kalmar, 2023-02-15:
@sunshinemyson , 
thanks it solves the failure.

closed 2023-02-15 by robert-kalmar