VeriSilicon/TIM-VX #586: TVM RPC error on Khadas VIM3 pro

fumao12345, 2023-05-15:
https://github.com/VeriSilicon/tvm/blob/vsi_npu/tests/python/contrib/test_vsi_npu/test_vsi_pytorch_model_all.py

I built and cross-compiled for Khadas VIM3 pro through tvm and TIM-VX 1.1.42 on x86_64 simulator. Running the test_vsi_tflite_model_all.py script to test mobilenet_v1_1.0_224_quant.tflite can be executed normally, but there are empty outputs when switching to other models.

**This is the output of running the quantized tflite model of InceptionNetV1.**
<details><summary> X86 Host</summary>
<p>

#productname=VSI SIMULATOR, pid=0x88
1. press any key and continue...
vsi_npu.py --> qnn.dequantize
vsi_npu.py --> nn.softmax
vsi_npu.py --> qnn.quantize
vsi_npu.py --> qnn.requantize
vsi_npu.py --> nn.avg_pool2d
vsi_npu.py --> qnn.requantize
vsi_npu.py --> qnn.requantize
vsi_npu.py --> qnn.requantize
vsi_npu.py --> qnn.requantize
vsi_npu.py --> qnn.requantize
vsi_npu.py --> qnn.requantize
vsi_npu.py --> qnn.requantize
vsi_npu.py --> qnn.requantize
vsi_npu.py --> qnn.requantize
vsi_npu.py --> qnn.requantize
vsi_npu.py --> qnn.requantize
vsi_npu.py --> qnn.requantize
vsi_npu.py --> nn.max_pool2d
vsi_npu.py --> nn.max_pool2d
vsi_npu.py --> qnn.requantize
vsi_npu.py --> qnn.requantize
vsi_npu.py --> qnn.requantize
vsi_npu.py --> qnn.requantize
vsi_npu.py --> qnn.requantize
vsi_npu.py --> nn.max_pool2d
vsi_npu.py --> qnn.concatenate
vsi_npu.py --> qnn.requantize
vsi_npu.py --> qnn.requantize
vsi_npu.py --> qnn.requantize
vsi_npu.py --> qnn.requantize
vsi_npu.py --> qnn.requantize
vsi_npu.py --> nn.max_pool2d
vsi_npu.py --> qnn.concatenate
vsi_npu.py --> nn.max_pool2d
vsi_npu.py --> qnn.requantize
vsi_npu.py --> qnn.requantize
vsi_npu.py --> qnn.requantize
vsi_npu.py --> qnn.requantize
vsi_npu.py --> qnn.requantize
vsi_npu.py --> nn.max_pool2d
vsi_npu.py --> qnn.concatenate
vsi_npu.py --> qnn.requantize
vsi_npu.py --> qnn.requantize
vsi_npu.py --> qnn.requantize
vsi_npu.py --> qnn.requantize
vsi_npu.py --> qnn.requantize
vsi_npu.py --> nn.max_pool2d
vsi_npu.py --> qnn.concatenate
vsi_npu.py --> qnn.requantize
vsi_npu.py --> qnn.requantize
vsi_npu.py --> qnn.requantize
vsi_npu.py --> qnn.requantize
vsi_npu.py --> qnn.requantize
vsi_npu.py --> nn.max_pool2d
vsi_npu.py --> qnn.concatenate
vsi_npu.py --> qnn.requantize
vsi_npu.py --> qnn.requantize
vsi_npu.py --> qnn.requantize
vsi_npu.py --> qnn.requantize
vsi_npu.py --> qnn.requantize
[... 460 lines cut ...]
VsiNpuModule::LoadFromBinary
LoadFromBinary: nbg size = 6862080
LoadFromBinary: input size = 1
LoadFromBinary: output size = 1
VsiNpuModule : DeSerializeTensorSpec
VsiNpuModule : DeSerializeTensorSpec2
VsiNpuModule : DeSerializeTensorSpec
VsiNpuModule : DeSerializeTensorSpec2
INFO:RPCServer:load_module /tmp/tmp2vjrwe64/lib.so
VsiNpuModule::GetFunction: _lookup_linked_param
VsiNpuModule::GetFunction: return early
VsiNpuModule::GetFunction: _lookup_linked_param
VsiNpuModule::GetFunction: return early
VsiNpuModule::GetFunction: _lookup_linked_param
VsiNpuModule::GetFunction: return early
VsiNpuModule::GetFunction: _lookup_linked_param
VsiNpuModule::GetFunction: return early
VsiNpuModule::GetFunction: tvmgen_default_vsi_npu_0
[     1] PLS isn't existed
Process Graph: 61439 ms or 61439858 us
VsiNpuModule::GetFunction: size: 2
INFO:RPCServer:Finish serving ('192.168.137.177', 41982)

</p>
</details>

sunshinemyson, 2023-05-23:
Thanks for report.

You can not get result for other tflite models because it is hang on device side. Could you share an example model which cannot get result?

fumao12345, 2023-05-23:
https://drive.google.com/drive/folders/1IYTqiebzVf_M0oUAjzG7BJ__iiC1P-Qm?usp=share_link

These three TFLite models produce empty output when executed with TVM RPC. Similarly, when I copy the model_lib.so generated by tvm's "lib.export_library(path_lib)" to Khadas VIM3 Pro and run the TVM runtime, I still encounter empty output.

fumao12345, 2023-05-23:
@sunshinemyson 
https://drive.google.com/drive/folders/1iX8GyEWZbzoAdiuW0AcU-4pApwhtldFp?usp=share_link

So far, I have tried executing MobileNetV2 and SqueezeNet TFLite models with TVM RPC or performing local compilation on Khadas VIM3 Pro, and both resulted in normal output.