VeriSilicon/TIM-VX #592: Provide ARM64 cmodel simulation tool

dataowner888, 2023-05-25:
Currently, you have released simulation tools under the X86_64 platform. Can you provide cmodel simulation tools under ARM64?

sunshinemyson, 2023-06-07:
@dataowner888 ,

Thanks for you are interest in our SW. Why do you want to ARM version？ From our POV, the simulator just enable you get a high-level understanding of our SW/HW capability, this is enough with X86 version. Could you explain more background for your usage?

Thanks

dataowner888, 2023-06-19:
Some AI frameworks have different processing methods under the ARM platform. When adapting to new NPU devices, it is necessary to repeatedly modify and debug the code. If ARM Cmodel tools are available, AI framework adaptation on real devices can be avoided and the adaptation progress can be accelerated.

sunshinemyson, 2023-06-26:
@dataowner888 ,

If you intend to debug high-level framework based on TIM-VX api, the x86 version have the same feature capability as ARM version.  could you give me more insight about "Some AI frameworks have different processing methods under the ARM platform."? According to our internal experience, we always develop the framework on x86 until we have a basic function coverage, then we switch to real device driver. We didn't see much limitation about this.

Thanks

closed 2023-08-08 by dataowner888