VeriSilicon/TIM-VX #630: Why is setting rounding mode invalid

dataowner888, 2023-08-08:
All operators (such as conv2d) convert floating-point to integer using the RTNE mode of rounding by default. I set conv2d to TO_ ZERO mode, there is no difference in the calculation results. Please help check the cause of the issue. The code is as follows:

`
TEST(Conv2d, mytest) {
  auto ctx = tim::vx::Context::Create();
  auto graph = ctx->CreateGraph();
  tim::vx::ShapeType input_shape({1846, 48, 3, 3});
  tim::vx::ShapeType weight_shape({3, 3, 3, 16});
  tim::vx::ShapeType bias_shape({16});
  tim::vx::ShapeType output_shape({923, 24, 16, 3});

  tim::vx::TensorSpec input_spec(tim::vx::DataType::INT8, input_shape,
                                 tim::vx::TensorAttribute::INPUT);
  //
  tim::vx::Quantization quantization_input;

  std::vector<float> scales_input = {0.007874};

quantization_input.SetType(tim::vx::QuantType::ASYMMETRIC);
  quantization_input.SetScales(scales_input);
  quantization_input.SetZeroPoints(std::vector<int32_t>(1, 0));
  //
  input_spec.SetQuantization(quantization_input);

  tim::vx::TensorSpec weight_spec(tim::vx::DataType::INT8, weight_shape,
                                  tim::vx::TensorAttribute::CONSTANT);
  //
  tim::vx::Quantization quantization_weight;
  std::vector<float> scales_weight(16);
  readFromFile((unsigned char *)(scales_weight.data()), 16*4,  (char *)"quant_scale_1.bin");
  quantization_weight.SetType(tim::vx::QuantType::SYMMETRIC_PER_CHANNEL);
  quantization_weight.SetChannelDim(3);
  quantization_weight.SetScales(scales_weight);
  quantization_weight.SetZeroPoints(std::vector<int32_t>(16, 0));
  //
  weight_spec.SetQuantization(quantization_weight);

  tim::vx::TensorSpec bias_spec(tim::vx::DataType::INT32, bias_shape,
                                tim::vx::TensorAttribute::CONSTANT);
  //
  tim::vx::Quantization quantization_bias;
  std::vector<float> scales_bias(16);
  readFromFile((unsigned char *)(scales_bias.data()), 16*4,  (char *)"quant_scale_2.bin");
  quantization_bias.SetType(tim::vx::QuantType::SYMMETRIC_PER_CHANNEL);
  quantization_bias.SetChannelDim(0);
  quantization_bias.SetScales(scales_bias);
  quantization_bias.SetZeroPoints(std::vector<int32_t>(16, 0));
  //

bias_spec.SetQuantization(quantization_bias);

  tim::vx::TensorSpec output_spec(tim::vx::DataType::INT8, output_shape,
                                                                tim::vx::TensorAttribute::OUTPUT);
  //
  tim::vx::Quantization quantization_output;

  std::vector<float> scales_output = {0.041692};

  quantization_output.SetType(tim::vx::QuantType::ASYMMETRIC);
  quantization_output.SetScales(scales_output);
  quantization_output.SetZeroPoints(std::vector<int32_t>(1, 0));
  //
  output_spec.SetQuantization(quantization_output);

printf("read data from file\n");

  std::vector<u_int8_t> input_data(input_shape[0]*input_shape[1]*input_shape[2]*input_shape[3]);
                readFromFile(input_data.data(), input_data.size(),  (char *)"conv2d_input.bin");
  std::vector<u_int8_t> weight_data(weight_shape[0]*weight_shape[1]*weight_shape[2]*weight_shape[3]);
  readFromFile(weight_data.data(), weight_data.size(),  (char *)"conv2d_weight.bin");
  std::vector<int32_t> bias_data(bias_shape[0]);
  readFromFile((unsigned char *)(bias_data.data()), bias_data.size()*4,  (char *)"conv2d_bias.bin");
  std::vector<u_int8_t> golden(output_shape[0]*output_shape[1]*output_shape[2]*output_shape[3]);

auto input_tensor = graph->CreateTensor(input_spec);
  auto weight_tensor = graph->CreateTensor(weight_spec, weight_data.data());
  auto bias_tensor = graph->CreateTensor(bias_spec, bias_data.data());
  auto output_tensor = graph->CreateTensor(output_spec);

  //auto padding = tim::vx::PadType::AUTO;
  std::array<uint32_t, 2> stride({2, 2});
  std::array<uint32_t, 2> dilation({1, 1});
  std::array<uint32_t, 4> padval({1,1,1,1});
  auto conv2d = graph->CreateOperation<tim::vx::ops::Conv2d>(
      padval, stride, dilation);

  (*conv2d)
      .BindInput(input_tensor)
      .BindInput(weight_tensor)
      .BindInput(bias_tensor)
      .BindOutput(output_tensor);

  conv2d->SetRoundingPolicy(tim::vx::OverflowPolicy::SATURATE, tim::vx::RoundingPolicy::TO_ZERO, tim::vx::RoundType::FLOOR);

  EXPECT_TRUE(graph->Compile());

  input_tensor->CopyDataToTensor(input_data.data());

  EXPECT_TRUE(graph->Run());

   uint32_t output_size = 1;
  for (auto i : output_tensor->GetShape()) {
    output_size *= i;
  }
  std::vector<u_int8_t> output(output_size);
  EXPECT_TRUE(output_tensor->CopyDataFromTensor(output.data()));

  writeToFile(output.data(), output.size(), (char *)"conv2d_output.bin");
 printf("status: ok\n");

  EXPECT_EQ(golden, output);
}`

dataowner888, 2023-08-09:
@sunshinemyson Can you check the issue list daily to respond to customer issues faster

sunshinemyson, 2023-08-09:
@dataowner888 ,

I'll check it with team. Thanks for raise your question.

sunshinemyson, 2023-08-11:
@dataowner888 ,

This configuration is not valid anymore, we can not change it from application level.

Thanks

dataowner888, 2023-08-11:
@sunshinemyson 
But I do need to change the floating-point rounding mode, and the OpenVX SPEC requires vxConvolutionLayer to support changing the floating-point rounding mode. Please provide me with a method to change the floating-point rounding mode. Thank you

dataowner888, 2023-08-15:
@sunshinemyson 
Are there any available methods?

sunshinemyson, 2023-08-17:
@dataowner888 ,

We don't have this feature. sorry for the inconvenience.

closed 2023-10-11 by sunshinemyson