VeriSilicon/TIM-VX #648: A311D 开发板运行，执行createGraph报错

zjd1988, 2023-09-15:
[image]
上图显示驱动版本和执行程序的保存信息，
下面是实际的log
```
pi@ubuntu:~/output$ dmesg | grep version
[    0.000000] Linux version 4.9.241-BPI-M2S (dangku@dangku-desktop) (gcc version 7.3.1 20180425 [linaro-7.3-2018.05 revision d29120a424ecfbc167ef90065c0eeb7f91977701] (Linaro GCC 7.3-2018.05) ) #1 SMP PREEMPT Fri Jun 17 12:23:43 HKT 2022
[    0.000000] Meson chip version = RevB (29:B - 10:0)
[    0.368307] vpu: driver version: v20190314(8-g12b)
[    0.558706] lcd: driver version: 20181012(6-g12b)
[    0.784547] squashfs: version 4.0 (2009/01/31) Phillip Lougher
[    0.786512] jffs2: version 2.2. (NAND) (SUMMARY)  © 2001-2006 Red Hat, Inc.
[    0.786879] fuse init (API version 7.26)
[    0.790837] Block layer SCSI generic (bsg) driver version 0.4 loaded (major 244)
[    0.827190] PPP generic driver version 2.4.2
[   10.459301] meson-mmc: mmc driver version: 3.02, 2017-05-15: New Emmc Host Controller
[   10.617451] aml_atvdemod_init: OK, atv demod version: V2.04.
[   10.868712] nf_conntrack version 0.5.0 (16384 buckets, 65536 max)
[   10.901092] registered taskstats version 1
[   10.909906] xhci-hcd xhci-hcd.0.auto: hcc params 0x0228fe64 hci version 0x110 quirks 0x00010010
[16288.075052] npu_version: 2
[16288.075392] Galcore version 6.4.8.7.1.1.1
pi@ubuntu:~/output$ ./model_verify --weight lenet_weight.bin --para lenet_graph.json --log_level 1
log level: 1
logger name: timvx_model
log file path:
[2023-09-15 11:40:21.776 +00:00] [timvx_model] [---D---] [thread 61303] [timvx_model.cpp TimVXModel 18] init logger success
[     1] Failed to open device: No such file or directory, Try again...
[     2] Failed to open device: No such file or directory, Try again...
[     3] Failed to open device: No such file or directory, Try again...
[     4] Failed to open device: No such file or directory, Try again...
[     5] _OpenDevice(1093): FATAL: Failed to open device, errno=No such file or directory.
```
手头使用的开发板是BPI 的M2S，芯片是A311D，使用的自己的仓库
https://github.com/zjd1988/timvx_engine.git 生成的可执行程序，从dmesg看，驱动已经加载了，但是依然提示找不到设备，这个报错是因为什么导致的呢？

zjd1988, 2023-09-20:
@sunshinemyson 这个UMD是指user mode driver吗，如何确认呢，是指libVSC.so这些库跟garlcore.ko是否配套吗？

sunshinemyson, 2023-09-23:
@zjd1988 ,

你的galcore.ko 是哪里拿到的呢？我也去其他途径问一下看有没有办法帮助你。

zjd1988, 2023-10-11:
在vim3 pro开发板可以加载模型和推理

closed 2023-10-11 by zjd1988