VeriSilicon/TIM-VX #654: Any plan to add quantized attention?

fengyuentau, 2023-10-25:
Hello guys, thanks for the great work! I wonder if you have any plan to support quantized attention?

chenfeiyue-cfy, 2023-10-26:
Hi, could you please give more information ? For example, what do you mean by "quantized attention"？Is it simply refers to quantizing all data types or other practices such as specific op quantization(e.g. softmax)?

fengyuentau, 2023-10-26:
Something like this: https://github.com/microsoft/onnxruntime/blob/main/docs/ContribOperators.md#com.microsoft.QAttention. Operator prototype may vary though.

chenfeiyue-cfy, 2023-10-26:
Thank you for your interest, unfortunately we have no plans to support this project yet.

fengyuentau, 2023-10-26:
Ok, thanks for the response. One more question is any ongoing plan about attention?

chenfeiyue-cfy, 2023-10-26:
It will depends on HW support.Which board (IP) are you currently using?

fengyuentau, 2023-10-26:
> Which board (IP) are you currently using?
Khadas VIM3 with A311D SoC.

chenfeiyue-cfy, 2023-10-27:
> > Which board (IP) are you currently using?
> 
> Khadas VIM3 with A311D SoC.
This hardware can provide functional support but cannot guarantee good performance such as accuracy and efficiency of GEMM operations in attention models.

fengyuentau, 2023-10-27:
Okay, thank you for your patience and all the helpful information.

closed 2023-10-27 by fengyuentau