VeriSilicon/TIM-VX #685: Amlogic A311D support for the latest kernel

serhii-matsyshyn, 2024-03-07:
Hello.
I want to use TIM-VX on the [Banana Pi BPI-CM4](https://wiki.banana-pi.org/Banana_Pi_BPI-CM4), which is based on the Amlogic A311D.   
TIM-VX is needed to use [NPU-accelerated OpenCV](https://github.com/opencv/opencv/wiki/TIM-VX-Backend-For-Running-OpenCV-On-NPU).

OS on the BPI-CM4 is the latest [Armbian 24.2.1 Bookworm with Linux 6.6.16-current-meson64](https://www.armbian.com/bananapicm4io/). The problem is that **galcore.ko**, which TIM-VX provides, is based on an old 4.9.0 kernel, and it is incompatible with the latest kernel. 

Is it worth trying to adapt some old, outdated kernel to run it on BPI-CM4 (which was stated as the only solution over three years ago #16), or is there a workaround for getting the up-to-date **galcore.ko** compatible with TIM-VX to use NPU (maybe you can give some tips on custom building; or provide pre-built **galcore.ko**)? 

Any help would be appreciated.

nikoladka, 2024-03-14:
Hi Serhii, I am trying to do the same, and my problem is even worse since I cannot even find  galcore.ko. I am using Debian 12 on my LibreCopmputer Alta with A311D. However I think  I have a clue what you need to do. I believe that  TIM-VX is little bit outdated, there is another effort is going on, which is Github Paddle Lite for A311D. They do have some instructions on how to deal with kernel. While TIM-VX is 2022 effort, the PAaddle Lite is rather 2023, so that is the hint.
While I am also chasing for OpenCV, I believe the first is better do the Paddle Lite, and then if success switch to OpenCV.
https://github.com/PaddlePaddle/FastDeploy/blob/develop/docs/en/build_and_install/a311d.md

nikoladka, 2024-04-25:
Hi Serhii, did you resolve the issue with Galcore?

sunshinemyson, 2024-04-28:
@serhii-matsyshyn ,

"The problem is that galcore.ko, which TIM-VX provides, is based on an old 4.9.0 kernel, and it is incompatible with the latest kernel."  TIM-VX doesn't provide galcore.ko or libOpenVX.so, you should get full driver sdk from SoC vendor(Amlogic A311D) then compile TIM-VX with their driver libs. 

We verified an out-of-date driver version from SoC, you can find it out from cmake/A311D.cmake - https://github.com/VeriSilicon/TIM-VX/blob/e1c2f0a18df7047feb6ee04187576906d10f5c87/cmake/A311D.cmake#L18C2-L18C104, however, we did not maintain the TIM-VX with out-of-date driver sdks. 

So, if you can get a full latest driver sdk from soc vendor, i think we can support you.

Thanks