VeriSilicon/TIM-VX #693: What does the producer and consumer mean?

engineer1109, 2024-04-12:
I found the graph interface has producer and consumer two means.
What mean?

And function:
```
UpdateTensorConsumersMap
RenewTensorConsumersMap
UpdateTensorProducerMap
GetConsumersOp
GetProducerOp
```
means what

chenfeiyue-cfy, 2024-04-24:
Hi, these map structures are desined to facilitate obtaining the entire topology of the graph from any location. The TensorProducerMap and TensorConsumerOp are both <tensor, op> type maps, where the former stores the successor operators for the current tensor, and the latter stores the predecessor operators for the current tensor.
As for functions, the Update*** functions are called when initializing the structure of the entire graph, meaning that every time UpdateMap is called, a new binding information between tensor and operation is added to the related map. RenewTensorConsumersMap is used to handle some const tensor  that are not supported by the definitions of operations in lower-level software. We convert the initialized timvx tensor into a new tensor through data conversion and other means, hence we need to modify the map. The Get*** function returns the smart pointer of the Consumer/Producer Operation for the current tensor.

engineer1109, 2024-04-24:
Thanks for your answer

closed 2024-04-24 by engineer1109