VeriSilicon/TIM-VX #704: Default layout inference pass

spacycoder, 2024-09-26:
Hi, I get these warnings when running an int8 quantized model on I.MX 8M Plus:

W [HandleLayoutInfer:332]Op 162: default layout inference pass.
W [HandleLayoutInfer:332]Op 56: default layout inference pass.

What do they mean? 

The model performs quite badly (its faster on CPU than the NPU)

sergiomsilva, 2024-11-12:
Hi @spacycoder , have you figured out? I'm having the same warnings and performance issues. It seems related to matmul operations inside a custom layer I have.

spacycoder, 2024-11-13:
No I haven't, but I also suspect it has something to do with matmul and possibly reshape ops

sunshinemyson, 2024-12-03:
@spacycoder  @sergiomsilva ,

"default layout inference pass" means that operation is not supported by layout inference framework - additional transpose operation may insert before/after your operation.

It could happen if the operation is customized op or reshape operation.

Thanks

BralSLA, 2024-12-05:
> @spacycoder @sergiomsilva ,
> 
> "default layout inference pass" means that operation is not supported by layout inference framework - additional transpose operation may insert before/after your operation.
> 
> It could happen if the operation is customized op or reshape operation.
> 
> Thanks

@sunshinemyson Currently facing the same issue. Is this something that can be solved using a different version of the vx_delegate? Would like to not rely on changing the model architecture if possible. Thanks.

hgaiser, 2024-12-16:
I got the exact same problem, a bunch of these warnings and the performance is worse compared to just CPU. Has anyone figured out what causes this?

sunshinemyson, 2025-04-17:
@BralSLA @hgaiser 

If you still have the problem, would you mind figure out what kind of layer raised "default layout inference pass"?

BralSLA, 2025-04-17:
@sunshinemyson Tracing this issue kept pointing to a split operation in our model. We ended up building against a different version of tflite. If you use the eIQ tool from NXP, or netron model viewer, you can search for the "Op" number in there. It was showing "Op: 162" corresponded with split.

sunshinemyson, 2025-06-11:
@GabrielGosden ，

Reshape operation is not supported in LayoutInference infrastructure. We don't have plan to support it. 

Thanks